fgcz/rawDiag #6: create order hexSticker

cpanse, 2018-04-05:
https://github.com/GuangchuangYu/hexSticker

cpanse, 2018-04-28:
# Install
```{r  eval = FALSE}
install.packages(c('scales', 'ggplotify'))
```
```{sh eval = FALSE}
sudo apt-get install libmagick++-dev
```

```{r  eval = FALSE}
devtools::install_github('GuangchuangYu/hexSticker')
```
# Have Fun

```{r}
library(hexSticker)
imgurl <- "https://github.com/protViz/rawDiag/blob/master/inst/shiny/demo/www/octopussy.png"
sticker("4Sticker2.png", package="rawDiag", p_size=14, s_x=1, s_y=1.1, s_width=.7, filename="sticker1.png", h_fill='lightgray', h_col = 'cornflowerblue', p_color = 'cornflowerblue', p_y = 0.35)

```

tobiasko, 2018-05-02:
[image]

tobiasko, 2018-05-02:
We should discuss the URL issue. It would be cool to have an url that does not contain "protViz". I think we should also add the F G C Z logo somehow.

tobiasko, 2018-05-24:
Stickers were ordered at stickermule

closed 2018-05-24 by tobiasko