fgcz/rawDiag #12: enable the vignette file for the package

cpanse, 2018-04-26:


cpanse, 2018-05-24:
devtools::install_github("protViz/rawDiag", build_vignettes = TRUE)

closed 2018-05-24 by cpanse

jjGG, 2018-05-24:
..
* checking for file
'/private/var/folders/83/dstdfqb54p36_whfvqrsqfkw0000gp/T/RtmpXwAq7E/devtools4bcf5efa30a4/protViz-rawDiag-971267d/DESCRIPTION'
... OK
* preparing 'rawDiag':
* checking DESCRIPTION meta-information ... OK
* installing the package to build vignettes
* creating vignettes ... ERROR
During startup - Warning messages:
1: Setting LC_CTYPE failed, using "C"
2: Setting LC_TIME failed, using "C"
3: Setting LC_MESSAGES failed, using "C"
4: Setting LC_MONETARY failed, using "C"
-- Attaching packages
----------------------------------------------------------------------------------
tidyverse 1.2.1 --
<U+221A> ggplot2 2.2.1     <U+221A> purrr   0.2.4
<U+221A> tibble  1.4.2     <U+221A> dplyr   0.7.4
<U+221A> tidyr   0.8.0     <U+221A> stringr 1.3.0
<U+221A> readr   1.1.1     <U+221A> forcats 0.3.0
-- Conflicts
-------------------------------------------------------------------------------------
tidyverse_conflicts() --
x dplyr::filter() masks stats::filter()
x dplyr::lag()    masks stats::lag()
Warning in readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file
'/private/var/folders/83/dstdfqb54p36_whfvqrsqfkw0000gp/T/RtmpQiQ25C/Rinst54672ffbd7d7/rawDiag/extdata/WU163763.RData',
probable reason 'No such file or directory'

Error: processing vignette 'rawDiag.Rnw' failed with diagnostics:
 chunk 7
Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
Execution halted

..

-- 

-----------------------------------------------------------------
Jonas Grossmann, Ph.D.
Functional Genomics Center Zurich (FGCZ)
University Zurich / ETH Zurich
Winterthurerstr. 190
Y32H88a
CH-8057 Zurich
Phone: +41-44-6353910
Fax:     +41-44-6353922
Office hours: Monday to Wednesday 09:00 -16:30
                   Thursday not available
                   Friday 09:00 -16:30
-----------------------------------------------------------------