fgcz/rawDiag #9: mzR reader method

cpanse, 2018-04-26:
to benefit from to the fantastic  PSI world.

wolski, 2018-04-26:
Map columns from RawDiag to CV terms. 

c("filename", "scanNumber", "StartTime", "BasePeakMass",
        "BasePeakIntensity", "TIC", "ScanType", "CycleNumber",
        "MSOrder", "MassAnalyzer", "PrecursorMass", "ChargeState",
        "IonInjectionTimems", "FTResolution", "MasterScanNumber",
        "LMCorrection", "ElapsedScanTimesec", "transient", "AGC",
        "AGCMode", "PrescanMode")

https://raw.githubusercontent.com/HUPO-PSI/psi-ms-CV/master/psi-ms.obo

tobiasko, 2018-04-26:
I will most likely not manage to do this before 15th of May (application deadline).

cpanse, 2018-04-27:
the code snippet below should do the job

```{r}
as.rawDiag.mzR <- function(object){
  #time.start <- Sys.time()
  runInfo <- runInfo(object)
  rv <- lapply(1:runInfo$scanCount, 
               function(x){data.frame(header(object, x))})
  rv <- bind_rows(rv)
  
  #time.end <- Sys.time()
  #message(time.start - time.end)
  
  rv <- rv[, c('acquisitionNum', 'retentionTime', 'basePeakMZ',
               'basePeakIntensity', 'totIonCurrent', 'msLevel', 'precursorMZ',
               'precursorCharge')]
  
  
  colnames(rv) <- c('scanNumber', 'StartTime', 'BasePeakMass',
                    'BasePeakIntensity', 'TIC', 'MSOrder', 'PrecursorMass',
                    'ChargeState')
  
  rv$filename <- basename(fileName(object))
  as.rawDiag(rv)
}
```

closed 2018-04-27 by cpanse