fgcz/rawDiag #36: change default path reading files

cpanse, 2018-10-24:
brought up by  Eduard

cpanse, 2018-10-25:
consider having a `rawDiagrc` in `$HOME` containing a list of directories;
 has to be checked on the windows platform

cpanse, 2018-10-25:
```{sh}
cp@vpn-global-dhcp2-191:~ > cat .Rprofile 
options(prompt = "R> ", continue = "+  ", width = 70, useFancyQuotes = FALSE)
local({
	r = getOption("repos")             
	r["CRAN"] = "https://stat.ethz.ch/CRAN/"
	options(repos = r)
})
rawDiagfilesystemRoot <- Sys.getenv('HOME')
rawDiagfilesystemDataDir = c("Downloads",
                        "data",
                        "raw"
  )
```

tobiasko, 2018-10-25:
The requested URL /~cpanse/rawDiag_0.0.18.tar.gz was not found on this server.

tobiasko, 2018-10-25:
tested for

`
> sessionInfo()
R version 3.3.2 (2016-10-31)
Platform: x86_64-apple-darwin13.4.0 (64-bit)
Running under: macOS  10.13.6

locale:
[1] C

attached base packages:
[1] parallel  stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] ggplot2_2.2.1   base64enc_0.1-3 rmarkdown_1.9   tidyr_0.8.0     shiny_1.1.0     rawDiag_0.0.18 
[7] protViz_0.3.1  

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.16     lubridate_1.7.3  lattice_0.20-35  assertthat_0.2.0 rprojroot_1.3-2 
 [6] digest_0.6.15    psych_1.8.3.3    mime_0.5         R6_2.2.2         cellranger_1.1.0
[11] plyr_1.8.4       backports_1.1.2  RSQLite_2.1.0    evaluate_0.10.1  httr_1.3.1      
[16] pillar_1.2.1     rlang_0.2.0      lazyeval_0.2.1   readxl_1.0.0     rstudioapi_0.7  
[21] hexbin_1.27.1    blob_1.1.1       readr_1.1.1      stringr_1.3.0    foreign_0.8-69  
[26] bit_1.1-12       munsell_0.4.3    broom_0.4.4      httpuv_1.4.5     modelr_0.1.1    
[31] pkgconfig_2.0.1  mnormt_1.5-5     htmltools_0.3.6  tibble_1.4.2     codetools_0.2-15
[36] crayon_1.3.4     dplyr_0.7.4      later_0.7.5      grid_3.3.2       nlme_3.1-130    
[41] jsonlite_1.5     xtable_1.8-2     gtable_0.2.0     DBI_0.8          magrittr_1.5    
[46] scales_0.5.0     stringi_1.1.7    reshape2_1.4.3   promises_1.0.1   bindrcpp_0.2.2  
[51] tidyverse_1.2.1  xml2_1.2.0       tools_3.3.2      forcats_0.3.0    bit64_0.9-7     
[56] glue_1.2.0       purrr_0.2.4      hms_0.4.2        yaml_2.1.18      colorspace_1.3-2
[61] rvest_0.3.2      memoise_1.1.0    knitr_1.20       bindr_0.1.1      haven_1.1.1
 `
works!