fgcz/rawDiag #37: install.R for windows

cpanse, 2018-10-29:
```{r}
#R
  
# Christian Panse <cp@fgcz.ethz.ch>
# Functional Genomics Center Zurich 2018

# System Requirements
pkgs <- c( 'devtools',
  'dplyr',
  'ggplot2',
  'hexbin',
  'magrittr',
  'parallel',
  'protViz',
  'rmarkdown',
  'RSQLite',
  'scales',
  'shiny',
  'tidyr',
  'tidyverse')

pkgs <- pkgs[(!pkgs %in% unique(installed.packages()[,'Package']))]
if(length(pkgs) > 0){install.packages(pkgs)}

# Installation
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.18.tar.gz')

# Testing
library(rawDiag)
(rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))
system.time(RAW <- read.raw(file = rawfile))
dim(RAW)
summary.rawDiag(RAW)
PlotScanFrequency(RAW)

# read all dimensions
dim(RAW)
RAW <- read.raw(file = rawfile, rawDiag = FALSE)
dim(RAW)

R.version.string; Sys.info()[c('sysname', 'version')]
```

jjGG, 2018-10-30:
hei, works on my mac without a problem. 
Only for XIC extraction (if I click "extract main peak") I get the error..
///////// console ///////
displays XICs of given masses.
arguments imply differing number of rows: 50, 0

jjGG, 2018-10-30:
Reopened on request of CP ;)