fgcz/rawDiag #46: R Rcpp <-> /clr:pure or /clr:safe C#

cpanse, 2018-12-07:
[image]

https://www.mono-project.com/docs/about-mono/languages/cplusplus/

https://www.mono-project.com/docs/faq/technical/
[image]

cpanse, 2019-05-28:
```{r}
#R

require(rDotNet)

## initialize
.cinit(dll="/Users/cp//__checkouts/R/rawDiag/exec/fgcz_raw.dll")

rawfile <- file.path(Sys.getenv("HOME"), "Downloads/20180220_14_autoQC01.raw")
## create an object and call a method

(obj <- .cnew ("Rawfile", rawfile))

obj$check()
obj$getFirstScanNumber()
L <- obj$GetSpectrum(2, "")

L[1]
L[2]
L[3]

print(L)
```

cpanse, 2019-05-28:
```
kill -9 `ps aux | grep CLRServer| awk '{print $2}' | tail -n 1 ` && ./build_linux_fgcz_raw.bash 
```

cpanse, 2019-05-28:
[image]

cpanse, 2019-05-28:
https://github.com/fgcz/rawDiag/blob/f9d53c0cb42f5a138fc3d9be8f2da625bb837619/src/fgcz_raw.cs#L265

cpanse, 2019-05-29:
# Install and test Rcode snippet

```{r}
install.packages('rDotNet')
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.35.tar.gz',  repo=NULL)
(rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))

S <- csReadScans(rawfile, 1:9)
S[1]
```

tobiasko, 2019-05-29:
Works!

```
> sessionInfo()
R version 3.5.3 (2019-03-11)
Platform: x86_64-apple-darwin15.6.0 (64-bit)
Running under: macOS Mojave 10.14.2

Matrix products: default
BLAS: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/3.5/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawDiag_0.0.35 rDotNet_0.9.1 

loaded via a namespace (and not attached):
 [1] tidyselect_0.2.5 purrr_0.3.2      haven_2.1.0      lattice_0.20-38  colorspace_1.4-1
 [6] generics_0.0.2   testthat_2.1.1   yaml_2.2.0       blob_1.1.1       rlang_0.3.2     
[11] hexbin_1.27.2    pillar_1.3.1     glue_1.3.1       DBI_1.0.0        tidyverse_1.2.1 
[16] bit64_0.9-7      modelr_0.1.4     readxl_1.3.1     protViz_0.4.0    plyr_1.8.4      
[21] stringr_1.4.0    munsell_0.5.0    gtable_0.2.0     cellranger_1.1.0 rvest_0.3.2     
[26] codetools_0.2-16 memoise_1.1.0    forcats_0.4.0    parallel_3.5.3   broom_0.5.1     
[31] Rcpp_1.0.1       readr_1.3.1      backports_1.1.3  scales_1.0.0     jsonlite_1.6    
[36] bit_1.1-14       ggplot2_3.1.0    hms_0.4.2        digest_0.6.18    stringi_1.4.3   
[41] dplyr_0.8.0.1    grid_3.5.3       tools_3.5.3      magrittr_1.5     lazyeval_0.2.2  
[46] tibble_2.1.1     RSQLite_2.1.1    crayon_1.3.4     tidyr_0.8.3      pkgconfig_2.0.2 
[51] xml2_1.2.0       lubridate_1.7.4  assertthat_0.2.1 httr_1.4.0       rstudioapi_0.10 
[56] R6_2.4.0         nlme_3.1-137     compiler_3.5.3  
```