fgcz/rawDiag #49: implement a read.raw.info method

cpanse, 2019-02-08:
```{r}
.read.raw.info <- function(file,
     mono = if(Sys.info()['sysname'] %in% c("Darwin", "Linux")) TRUE else FALSE,
     exe = file.path(path.package(package = "rawDiag"), "exec", "fgcz_raw.exe"),
     mono_path = "",
     argv = "info",
     system2_call = TRUE,
     method = "thermo"){

  if(system2_call && method == 'thermo'){

    tf <- tempfile(fileext = '.tsv')
    tf.err <- tempfile(fileext = '.tsv')

    message(paste("system2 is writting to tempfile ", tf, "..."))

    if (mono){
      rvs <- system2("mono", args = c(exe, shQuote(file), argv),
                     stdout = tf)
    }else{
      rvs <- system2(exe, args = c(shQuote(file), argv),
                     stderr = tf.err,
                     stdout = tf)
    }

    if (rvs == 0){
      rv <- read.csv(tf,  sep = ":",   stringsAsFactors = TRUE, header = FALSE,
                     col.names = c('attribute', 'value'))

      message(paste("unlinking", tf, "..."))

      unlink(tf)
      # unlink(tfstdout)
      return(rv)
    }
  }
  NULL
}
```

cpanse, 2019-02-08:
https://github.com/fgcz/rawDiag/blob/c50ff4079aad823aa94ffae8639bbc2d7c91ddb0/R/rawDiag.R#L524

kaniawave, 2019-02-08:
It works! Confirmed! Thanks for the great piece of code.

cpanse, 2019-05-28:
[image]

@kaniawave since version 0.0.34 there is a rawDiag::read.raw.info function.

```{r}
(rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))
Y <- read.raw.info(rawfile)
```

commit f0540542 (2019-05-28): resolved #49
commit 757a6ed4 (2019-05-28): resolved #49 @tobiasko
closed 2019-05-28 by cpanse