fgcz/rawDiag #52: Add Signal to Noise data to ReadScan

David-Degnan, 2019-07-31:
Hello! Thank you for developing this useful package. Is there any way to add the signal to noise info to the object returned by ReadScans? This is a value that Thermo stores in the .raw file for every peak, besides the m/z and intensity. I believe that it can be obtained from one of the ThermoFisher.CommonCore DLLs already utilized by rawDiag. Thanks!

tobiasko, 2019-08-27:
Hi David,

to my knowledge S/N is not something that gets written into the raw file, but is rather estimated for each peak by an offline algorithm that examines the local surrounding. Where does your info come from that it is written into the file?

Best,
Tobi

David-Degnan, 2019-08-30:
Thanks for the reply, Tobi!

You are correct. We need the noise or background information stored for each peak.

closed 2019-08-30 by tobiasko

David-Degnan, 2019-08-30:
Hello Tobi, 

It seems I may have been unclear. We would still like to add signal/background information to the object returned by ReadScans. We believe this value is stored in one of the ThermoFisher DLLs, likely ThermoFisher.CommonCore.RawFileReader.dll. Here's a reference: https://github.com/PNNL-Comp-Mass-Spec/Thermo-Raw-File-Reader/blob/master/XRawFileIO.cs#L2508

Thank you!

tobiasko, 2019-09-02:
Hi David,

in general DLLs do not store any values:

https://support.microsoft.com/en-us/help/815065/what-is-a-dll

In short, DLLs are code modules. The code line you linked above:

```var data = mXRawFile.GetCentroidStream(scan, false);```

references the method ```GetCentroidStream``` and to my knowledge it returns a set of position and intensity values in array format. Position is equivalent to m/z and intensity is the abs. abundance aka the signal. Noise is not stored in a raw file can can be estimated by analysing the local/global environment of any signal.

@cpanse Is this correct?

Best,
Tobi

tobiasko, 2019-09-02:
btw: I guess the only person that really KNOWs what is inside raw files is jim.Shofstahl@thermofisher.com
I could also be that the [.NET assemblies](https://planetorbitrap.com/rawfilereader) do not make it accessible. :-(

David-Degnan, 2019-09-10:
Hello Tobi, 

The GetCentroidStream returns an array of position, intensity, resolutions, baselines, noises, and charges. We would like the noises added to ReadScans. https://github.com/PNNL-Comp-Mass-Spec/Thermo-Raw-File-Reader/blob/master/XRawFileIO.cs#L2520 

Thank you!

-David