fgcz/rawDiag #55: read.raw output

viviantra, 2020-04-25:
Hi, I have a question about read.raw.  

I was able to install the package.  However, I am not sure if I have got the expected information from read.raw function.  Can someone verify my output is correct?  What I don't understand is: why am I only getting length of 573?  More specifically, I would expect more than one entry in each ScanNumber.  For example, there are in reality multiple ions in scan#1 (full scan), but there is only Row #1 corresponding to scanNumber == "1", with mass 445.1181. 

(rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))
system.time(RAW <- read.raw(file = rawfile))
dim(RAW)
summary(RAW)

**My output:**
> dim(RAW)
[1] 573  21
> summary(RAW)
   filename           scanNumber    StartTime        BasePeakMass   BasePeakIntensity       TIC          
 Length:573         Min.   :  1   Min.   :0.00162   Min.   :216.1   Min.   :    4768   Min.   :   23224 
 Class :character   1st Qu.:144   1st Qu.:0.19741   1st Qu.:216.4   1st Qu.:   10296   1st Qu.:   51955 
 Mode  :character   Median :287   Median :0.39117   Median :303.5   Median :   22549   Median :   80879 
                    Mean   :287   Mean   :0.39184   Mean   :296.3   Mean   :  352735   Mean   : 4032142 
                    3rd Qu.:430   3rd Qu.:0.58665   3rd Qu.:303.5   3rd Qu.:   35850   3rd Qu.:  162624 
                    Max.   :573   Max.   :0.78011   Max.   :696.0   Max.   :16794288   Max.   :97975704 
                                                                                                         
   ScanType          CycleNumber      MSOrder          MassAnalyzer       PrecursorMass        AGC          
 Length:573         Min.   : 0.00   Length:573         Length:573         Min.   : 487.3   Length:573       
 Class :character   1st Qu.: 6.00   Class :character   Class :character   1st Qu.: 582.3   Class :character 
 Mode  :character   Median :13.00   Mode  :character   Mode  :character   Median : 669.8   Mode  :character 
                    Mean   :12.52                                         Mean   : 663.3                    
                    3rd Qu.:19.00                                         3rd Qu.: 722.3                    
                    Max.   :26.00                                         Max.   :1075.0                    
                                                                                                             
  ChargeState    IonInjectionTimems  FTResolution     LMCorrection  PrescanMode    MasterScanNumber ElapsedScanTimesec
Min.   :0.000   Min.   :39.04      Min.   : 30000   Min.   :0     Min.   :-21.0   Min.   :  1      Min.   :0.06875  
 1st Qu.:2.000   1st Qu.:55.00      1st Qu.: 30000   1st Qu.:0     1st Qu.:-16.0   1st Qu.:133      1st Qu.:0.06912  
 Median :2.000   Median :55.00      Median : 30000   Median :0     Median :-10.0   Median :276      Median :0.06925  
 Mean   :1.997   Mean   :54.76      Mean   : 34241   Mean   :0     Mean   :-10.3   Mean   :276      Mean   :0.08166  
 3rd Qu.:2.000   3rd Qu.:55.00      3rd Qu.: 30000   3rd Qu.:0     3rd Qu.: -4.0   3rd Qu.:419      3rd Qu.:0.06938  
 Max.   :3.000   Max.   :55.00      Max.   :120000   Max.   :0     Max.   :  1.0   Max.   :551      Max.   :0.25319  
                                                                                   NA's   :27       NA's   :1        
 AGCMode          transient    
 Mode:logical   Min.   : 64.00 
 NA's:573       1st Qu.: 64.00 
                Median : 64.00 
                Mean   : 73.05 
                3rd Qu.: 64.00 
                Max.   :256.00

tobiasko, 2020-04-26:
Hi Vivian,

the test data of the package only contains 573 scans. Please check by using Thermo QualBrowser or Freestyle. The raw file is located in a subfolder of the package. To locate the package folder use:

```
> file.path(path.package(package = 'rawDiag'))
[1] "/Library/Frameworks/R.framework/Versions/3.6/Resources/library/rawDiag"
```
Go to the package folder and change to the `extdata` subfolder. You will find `sample.raw`.

`read.raw()` will return a data.frame including one row for each scan. The columns are descriptive variables describing properties of each scan.  We like to refer to these attributes as scan-level metadata. `read_raw()` does **not** return peak lists as you suspected! The atrributes are pre-filtered. If you want to see the full range of attributes as set by Thermo use:

```
> str(read.raw(rawfile, rawDiag = FALSE))
executing mono /Library/Frameworks/R.framework/Versions/3.6/Resources/library/rawDiag/exec/fgcz_raw.exe /Library/Frameworks/R.framework/Versions/3.6/Resources/library/rawDiag/extdata/sample.raw qc ...
Classes ‘rawDiag’ and 'data.frame':	573 obs. of  82 variables:
 $ filename              : chr  "sample.raw" "sample.raw" "sample.raw" "sample.raw" ...
 $ scanNumber            : int  1 2 3 4 5 6 7 8 9 10 ...
 $ ScanEventNumber       : int  0 1 1 1 1 1 1 1 1 1 ...
 $ StartTime             : num  0.00162 0.00584 0.00699 0.00815 0.0093 ...
 $ BasePeakMass          : num  445 303 303 216 303 ...
 $ BasePeakIntensity     : num  5979308 9080 9849 28202 7352 ...
 $ TIC                   : num  78725448 41041 53557 188951 57692 ...
 $ ScanType              : chr  "FTMS + c NSI Full ms [350.0000-1800.0000]" "FTMS + c NSI Full ms2 487.2567@hcd28.00 [140.0000-1015.0000]" "FTMS + c NSI Full ms2 644.8226@hcd28.00 [140.0000-1335.0000]" "FTMS + c NSI Full ms2 683.8279@hcd28.00 [140.0000-1415.0000]" ...
 $ CycleNumber           : int  0 0 0 0 0 0 0 0 0 0 ...
 $ Frequency             : int  0 0 0 0 0 0 0 0 0 0 ...
 $ HighMass              : int  1800 1015 1335 1415 1135 1385 1415 1445 1500 1290 ...
 $ IonizationMode        : chr  "NanoSpray" "NanoSpray" "NanoSpray" "NanoSpray" ...
 $ MSOrder               : chr  "Ms" "Ms2" "Ms2" "Ms2" ...
 $ MassAnalyzer          : chr  "MassAnalyzerFTMS" "MassAnalyzerFTMS" "MassAnalyzerFTMS" "MassAnalyzerFTMS" ...
 $ Detector              : chr  "Any" "Any" "Any" "Any" ...
 $ Lock                  : chr  "Any" "Any" "Any" "Any" ...
 $ PrecursorMass         : num  1075 487 645 684 547 ...
 $ LastPrecursorMass     : int  1800 0 0 0 0 0 0 0 0 0 ...
 $ CollisionEnergy       : int  0 28 28 28 28 28 28 28 28 28 ...
 $ IsolationWidth        : num  1450 1.4 1.4 1.4 1.4 ...
 $ MultipleInjection     : chr  "ii" "h" "h" "h" ...
 $ MultiInjectInfo       : chr  "IT=28;28" "IT=55" "IT=55" "IT=55" ...
 $ AGC                   : chr  "On" "On" "On" "On" ...
 $ MicroScanCount        : int  1 1 1 1 1 1 1 1 1 1 ...
 $ ScanSegment           : int  1 1 1 1 1 1 1 1 1 1 ...
 $ ScanEvent             : int  1 2 2 2 2 2 2 2 2 2 ...
 $ MasterIndex           : int  0 0 0 0 0 0 0 0 0 0 ...
 $ ChargeState           : int  0 2 2 2 2 2 2 2 2 2 ...
 $ MonoisotopicmZ        : num  0 0 0 0 0 0 0 0 0 0 ...
 $ IonInjectionTimems    : num  55 55 55 55 55 55 55 55 55 55 ...
 $ MaxIonTimems          : num  55 55 55 55 55 55 55 55 55 55 ...
 $ FTResolution          : int  120000 30000 30000 30000 30000 30000 30000 30000 30000 30000 ...
 $ MS2IsolationWidth     : num  1450 1.4 1.4 1.4 1.4 1.4 1.4 1.4 1.4 1.4 ...
 $ MS2IsolationOffset    : num  0 0 0 0 0 0 0 0 0 0 ...
 $ AGCTarget             : int  3000000 100000 100000 100000 100000 100000 100000 100000 100000 100000 ...
 $ HCDEnergy             : num  NA 28 28 28 28 28 28 28 28 28 ...
 $ AnalyzerTemperature   : num  27.1 27.1 27.1 27.1 27.1 ...
 $ MassCalibration       : logi  NA NA NA NA NA NA ...
 $ ConversionParameterB  : num  2.12e+08 2.12e+08 2.12e+08 2.12e+08 2.12e+08 ...
 $ ConversionParameterC  : num  -41163068 -41163399 -41163399 -41163397 -41163399 ...
 $ TemperatureCompppm    : num  -2.35 -2.35 -2.35 -2.35 -2.35 -2.35 -2.35 -2.35 -2.35 -2.35 ...
 $ RFCompppm             : num  -0.09 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 ...
 $ SpaceChargeCompppm    : num  -2.3 -0.01 -0.01 -0.02 -0.01 -0.04 -0.02 -0.01 -0.02 -0.01 ...
 $ ResolutionCompppm     : num  0.57 0.19 0.19 0.19 0.19 0.19 0.19 0.19 0.19 0.19 ...
 $ NumberofLockMasses    : int  0 0 0 0 0 0 0 0 0 0 ...
 $ LockMass1mZ           : num  0 0 0 0 0 0 0 0 0 0 ...
 $ LockMass2mZ           : num  0 0 0 0 0 0 0 0 0 0 ...
 $ LockMass3mZ           : num  0 0 0 0 0 0 0 0 0 0 ...
 $ LMSearchWindowppm     : num  0 0 0 0 0 0 0 0 0 0 ...
 $ LMSearchWindowmmu     : num  0 0 0 0 0 0 0 0 0 0 ...
 $ NumberofLMFound       : int  0 0 0 0 0 0 0 0 0 0 ...
 $ LastLockingsec        : num  0 0 0 0 0 0 0 0 0 0 ...
 $ LMmZCorrectionppm     : num  0 0 0 0 0 0 0 0 0 0 ...
 $ IonOpticsSettings     : logi  NA NA NA NA NA NA ...
[... 10 lines cut ...]
 $ IntensCompFactor      : num  1.428 0.554 0.554 0.56 0.555 ...
 $ ResDepIntens          : num  0.981 1 1 2.466 1 ...
 $ CTCDNumF              : int  0 0 0 0 0 0 0 0 0 0 ...
 $ CTCDComp              : num  1.01 1.01 1.01 1.01 1.01 ...
 $ CTCDScScr             : num  0.844 0 0 0 0 0 0 0 0 0 ...
 $ RawOvFtT              : num  2378646 11139 10542 21162 12215 ...
 $ LCFWHMparameter       : num  15 15 15 15 15 15 15 15 15 15 ...
 $ Rod                   : int  0 0 0 0 0 0 0 0 0 0 ...
[... cut ...]
 $ HCDEnergyeV           : num  0 24.6 32.5 34.5 27.6 ...
 $ AGCFill               : num  0.76 0.03 0.03 0.03 0.04 0.39 0.05 0.03 0.03 0.03 ...
 $ Injectiont0           : num  -0.071 0 0 0 0 0 0 0 0 0 ...
 $ t0FLP                 : num  355 0 0 0 0 ...
 $ AccessId              : int  0 0 0 0 0 0 0 0 0 0 ...
 $ AnalogInput1V         : num  0 0 0 0 0 0 0 0 0 0 ...
 $ AnalogInput2V         : num  0 0 0 0 0 0 0 0 0 0 ...
> 
```

The list of attributes will change by instrument used for data recording.

Best,
Tobi

tobiasko, 2020-04-26:
For the extraction of peak lists we have implemented `readSan()`

```
> rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw')
> rawfile
[1] "/Library/Frameworks/R.framework/Versions/3.6/Resources/library/rawDiag/extdata/sample.raw"
> readScans(rawfile = rawfile, 1)
[[1]]
$scan
[1] 1

$scanType
[1] "FTMS + c NSI Full ms [350.0000-1800.0000]"

$rtinseconds
[1] 0.097

$pepmass
[1]    1075 5979308

$centroidStream
[1] TRUE

$title
[1] "File: sample.raw; SpectrumID: ; scans: 1"

$charge
[1] 0

$monoisotopicMz
[1] 0

$mZ
  [1]  352.2367  352.3192  352.3641  353.1436  353.2399  353.2521  353.2821  353.3031
  [9]  353.3180  353.3397  353.3763  354.2429  354.3435  354.3795  355.0683  355.1405
 [17]  355.1895  355.2107  355.2223  355.2286  355.2467  355.2828  355.2977  355.3191
 [25]  355.3342  355.3554  356.0674  356.2779  356.2853  356.3011  356.3219  356.3503
 [33]  357.0113  357.0477  357.0653  357.0715  357.1997  357.2367  357.2617  357.2809
 [41]  357.2982  357.3135  358.0101  358.0464  358.0648  358.0696  358.1663  358.2399
 [49]  358.3012  358.3662  358.9902  359.0268  359.0440  359.0620  359.0685  359.2410
 [57]  359.2561  359.2714  359.2779  359.2929  359.3143  359.3291  359.3507  360.0257
 [65]  360.0440  360.0617  360.2964  360.3180  360.9875  361.0239  361.0308  361.1456
 [73]  361.2204  361.2349  361.2932  361.3296  361.3446  362.0234  362.2097  362.2241
 [81]  362.3329  362.3479  363.0209  363.1703  363.2143  363.2259  363.2514  363.2879
 [89]  363.3005  363.3087  363.3242  363.3365  363.3600  364.1737  364.2547  364.3130
 [97]  364.3404  364.3647  365.2151  365.2520  365.2671  365.3037  365.3191  365.3399
[105]  365.3759  366.3797  367.0684  367.2676  367.2982  367.3344  367.3557  367.3921
[113]  368.0679  368.1049  368.2716  368.3016  368.3374  368.3953  369.0660  369.0851
[121]  369.1202  369.2986  369.3138  369.3345  369.3501  369.3712  370.1193  370.1261
[129]  370.3533  370.3746  371.0636  371.0996  371.3139  371.3767  372.0621  372.0988
[137]  372.2546  372.2964  372.3172  373.0424  373.0605  373.0789  373.0965  373.1026
[145]  373.1490  373.1840  373.2574  373.3083  373.3205  373.3275  373.3442  373.3816
[153]  373.8855  374.0410  374.0599  374.0773  374.0960  374.1007  374.3117  374.3215
[161]  375.0395  375.0570  375.0757  375.0822  375.0934  375.0997  375.2516  375.3088
[169]  375.3239  375.3453  375.3603  376.0754  376.0922  376.3116  376.3483  377.2303
[177]  377.2494  377.2667  377.3032  377.3177  377.3245  377.3399  377.3523  377.3761
[185]  378.0018  378.2704  378.2830  378.3065  378.3282  378.3564  379.2366  379.3917
[193]  380.3958  381.2980  381.3346  381.3703  381.4071  382.4111  383.1393  383.2201
[201]  383.3131  383.3291  383.3498  383.3863  384.3903  384.8052  385.0483  385.0790
[209]  385.1153  385.1517  385.3299  385.3447  386.1503  386.3328  386.3481  386.3975
[217]  386.8035  387.0458  387.0945  387.1312  387.1487  387.3608  387.4012  388.0934
[225]  388.1259  388.3406  388.3475  388.8012  389.0237  389.0737  389.0916  389.0981
[233]  389.1101  389.1262  389.2670  389.3246  389.3443  389.3610  390.0901  390.1092
[241]  390.1224  390.3643  390.7993  391.0536  391.1079  391.1133  391.1375  391.2016
[249]  391.2824  391.3397  391.3909  392.2858  392.3434  393.2888  393.3706  393.4066
[257]  394.2921  394.4107  395.2619  395.3499  395.3867  395.4236  396.3544  396.4263
[265]  397.2679  397.3301  397.3657  397.4024  398.4052  399.0937  399.2487  399.2641
[273]  399.2721  399.2839  399.3239  399.3452  400.3035  400.3487  400.9825  401.1462
[281]  401.2511  401.3400  401.3521  401.3756  402.1784  403.1253  403.2309  403.3034
[289]  403.3401  404.1254  404.1569  404.2343  405.1048  405.1227  405.2369  405.2620
[... 161 lines cut ...]
[619]  467290.500    6396.663  141077.781   55548.367    7640.772    3064.441
[625]  157427.766   88389.172   27174.506   58558.859   15813.728  155376.922
[631]   22402.613    9336.749    3934.167   87670.102    7271.868    3445.736
[637]   66527.602   19200.289    6895.857    9039.146    9262.235    4585.278
[643]    8994.686    7196.707  153635.672   98957.516   35858.461   68705.930
[649]   15144.061   30810.160   11121.040   13706.985    6242.925  953891.750
[655]  639480.938  473981.562  221084.500   89429.172   28657.117    9964.812
[661]   51669.785   30503.070    5294.802   15764.815   55970.824    9280.074
[667]   29152.461    4862.654   24736.873   12620.339    3971.855    2874.574
[673]    3432.849    4561.159    4856.255    6829.927    3820.718   22921.582
[679]   18850.404   21227.533   12546.057   12709.188    4536.234    9908.554
[685]    4353.638  289258.656  219174.844  168046.875   78459.969   34666.238
[691]   10554.942    4697.209   13037.872   13381.538   10005.523   11148.404
[697]    5884.552    9210.484    7488.578    3427.392    4928.978    5248.564
[703]    3864.771    4030.701    3750.559   11002.332   11549.372   10505.842
[709]    7954.002    3595.303   76288.188   66279.039   56249.855   25661.854
[715]   16712.242    6938.754    2601.111    3051.912    2967.207    2966.276
[721]    2947.383    3737.897    4220.827    6127.435   20817.350   12836.690
[727]   20816.096    8052.212    2881.511    2622.339    4852.667    3100.941
[733]    2476.037    2490.257    2733.165    2779.918    2794.363    2531.890

attr(,"class")
[1] "list"     "peaklist"

```

tobiasko, 2020-04-26:
Details on using `readScans()` and `readXICs` are shown [here](http://fgcz-ms.uzh.ch/~cpanse/rawDiagXICdemo.pdf)

viviantra, 2020-04-26:
Thank you!  Very helpful! 

I think readXICs is what I need.  If I want to extract peptides intensities from a batch of files, how do I use readXICs across multiple files in the same directory?  Thank you!

tobiasko, 2020-04-27:
Hi Vivian,

`readXICs()` is indeed not vectorised. You would need to use a function from the `apply()` family to do this. Our FAQ section lists an example using a multi core apply control structure, see [here](https://github.com/fgcz/rawDiag#341-i-would-like-to-load-multiple-files-into-a-single-dataframe-to-do-comparisons-what-is-the-preferred-method-for-doing-so). That should also work for `readXICs`. The `mclapply()` function has been moved to the `parallel` package. Use
```
library(parallel)
```
to make it available.

Best,
Tobi

tobiasko, 2020-04-27:
You might also check the [https://github.com/compomics/ThermoRawFileParser](https://github.com/compomics/https://github.com/compomics/ThermoRawFileParser) XIC function.

viviantra, 2020-05-01:
Hi Tobi, 

Yes readXICs() is exactly what I need!!  

I was able to extract all the features from all files and analyze them together.  The mass column from readXICs() is the mass used for extraction.  It would be nice to list the mass in the raw feature as another column.  I have not tried readScans().  I imagine readScans will have the actual mass. 

Thank you so much for the help and information.  The tool has helped me tremendously!

Vivian

closed 2020-05-01 by viviantra

tobiasko, 2020-05-01:
Hi Vivian,

nice to hear that our package was useful to you. Regarding "the mass": Not sure if I get you correctly, but we list the `PrecursorMass` in the raw data.frame. For MS2 scans this is the m/z values used for quadrupole isolation.

Best,
Tobi