fgcz/rawDiag #56: test rawDiag release on R-4.0.0

cpanse, 2020-04-27:
!! Linux

!! Microsoft Windows 2012

```
> library(rawDiag)
Package 'rawDiag' version 0.0.34
> (rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.$
[1] "C:/Program Files/R/R-4.0.0/library/rawDiag/extdata/sample.raw"
> system.time(RAW <- read.raw(file = rawfile))
system2 is writting to tempfile  C:\Users\cpanse\AppData\Local\Temp\Rtmpy8y7pQ\f
ilea05828cf7dcbtsv ...
unlinking C:\Users\cpanse\AppData\Local\Temp\Rtmpy8y7pQ\filea05828cf7dcbtsv ...
MasterScanNumber calculated
renamed LMmZCorrectionppm to LMCorrection
renamed AGCPSMode to PrescanMode
   user  system elapsed
   0.13    0.00    0.77
Warning message:
In is.rawDiag(object) :
  missing column name(s): MasterScanNumber, LMCorrection, ElapsedScanTimesec, tr
ansient, AGCMode, PrescanMode
> head(RAW)
    filename scanNumber   StartTime BasePeakMass BasePeakIntensity         TIC
1 sample.raw          1 0.001619751     445.1181       5979307.500 78725448.00
2 sample.raw          2 0.005839446     303.4642          9080.061    41040.95
3 sample.raw          3 0.006989346     303.4531          9849.274    53556.81
4 sample.raw          4 0.008149761     216.2900         28202.479   188950.88
5 sample.raw          5 0.009297695     303.4684          7351.867    57691.80
6 sample.raw          6 0.010449778     361.0250         86448.688   551090.31
                                                      ScanType CycleNumber
1                    FTMS + c NSI Full ms [350.0000-1800.0000]           0
2 FTMS + c NSI Full ms2 487.2567@hcd28.00 [140.0000-1015.0000]           0
3 FTMS + c NSI Full ms2 644.8226@hcd28.00 [140.0000-1335.0000]           0
4 FTMS + c NSI Full ms2 683.8279@hcd28.00 [140.0000-1415.0000]           0
5 FTMS + c NSI Full ms2 547.2980@hcd28.00 [140.0000-1135.0000]           0
6 FTMS + c NSI Full ms2 669.8381@hcd28.00 [140.0000-1385.0000]           0
  MSOrder     MassAnalyzer PrecursorMass AGC ChargeState IonInjectionTimems
1      Ms MassAnalyzerFTMS     1075.0000  On           0                 55
2     Ms2 MassAnalyzerFTMS      487.2567  On           2                 55
3     Ms2 MassAnalyzerFTMS      644.8226  On           2                 55
4     Ms2 MassAnalyzerFTMS      683.8279  On           2                 55
5     Ms2 MassAnalyzerFTMS      547.2980  On           2                 55
6     Ms2 MassAnalyzerFTMS      669.8381  On           2                 55
  FTResolution LMCorrection PrescanMode MasterScanNumber ElapsedScanTimesec
1       120000            0           1               NA         0.25318167
2        30000            0          -1                1         0.06899404
3        30000            0          -2                1         0.06962491
4        30000            0          -3                1         0.06887603
5        30000            0          -4                1         0.06912496
6        30000            0          -5                1         0.06912612
  AGCMode transient
1      NA       256
2      NA        64
3      NA        64
4      NA        64
5      NA        64
6      NA        64
> sessionInfo()
R version 4.0.0 (2020-04-24)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server 2012 R2 x64 (build 9600)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252
[2] LC_CTYPE=English_United States.1252
[3] LC_MONETARY=English_United States.1252
[4] LC_NUMERIC=C
[5] LC_TIME=English_United States.1252

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base

other attached packages:
[1] rawDiag_0.0.34 protViz_0.6.4

loaded via a namespace (and not attached):
 [1] tidyselect_1.0.0 purrr_0.3.4      haven_2.2.0      lattice_0.20-41
 [5] colorspace_1.4-1 vctrs_0.2.4      generics_0.0.2   yaml_2.2.1
 [9] blob_1.2.1       rlang_0.4.5      hexbin_1.28.1    pillar_1.4.3
[13] glue_1.4.0       DBI_1.1.0        tidyverse_1.3.0  bit64_0.9-7
[17] dbplyr_1.4.3     modelr_0.1.6     readxl_1.3.1     lifecycle_0.2.0
[21] stringr_1.4.0    munsell_0.5.0    gtable_0.3.0     cellranger_1.1.0
[25] rvest_0.3.5      codetools_0.2-16 memoise_1.1.0    forcats_0.5.0
[29] parallel_4.0.0   fansi_0.4.1      broom_0.5.6      Rcpp_1.0.4.6
[33] readr_1.3.1      scales_1.1.0     backports_1.1.6  jsonlite_1.6.1
[37] fs_1.4.1         bit_1.1-15.2     ggplot2_3.3.0    hms_0.5.3
[41] digest_0.6.25    stringi_1.4.6    dplyr_0.8.5      grid_4.0.0
[45] cli_2.0.2        tools_4.0.0      magrittr_1.5     tibble_3.0.0
[49] RSQLite_2.2.0    crayon_1.3.4     tidyr_1.0.2      pkgconfig_2.0.3
[53] ellipsis_0.3.0   xml2_1.3.1       reprex_0.3.0     lubridate_1.7.8
[57] assertthat_0.2.1 httr_1.4.1       R6_2.4.1         nlme_3.1-147
[61] compiler_4.0.0
>
```

!! macOS 10.15.4 - Darwin 19.4.0

viviantra, 2020-05-24:
I could not use plot(XIC)

> sessionInfo()
R version 4.0.0 (2020-04-24)
Platform: x86_64-apple-darwin17.0 (64-bit)
Running under: macOS Catalina 10.15.4

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/4.0/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] parallel  stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:

other attached packages:
 [1] factoextra_1.0.7 gridExtra_2.3    tidyr_1.0.3      stringr_1.4.0    ggplot2_3.3.0    dplyr_0.8.5      mzR_2.22.0       Rcpp_1.0.4.6    
 [9] devtools_2.3.0   usethis_1.6.1    rawDiag_0.0.32   protViz_0.6.8   

loaded via a namespace (and not attached):
 [1] Biobase_2.48.0      httr_1.4.1          pkgload_1.0.2       bit64_0.9-7         jsonlite_1.6.1      modelr_0.1.7       
 [7] assertthat_0.2.1    blob_1.2.1          cellranger_1.1.0    ggrepel_0.8.2       remotes_2.1.1       sessioninfo_1.1.1  
[13] pillar_1.4.4        RSQLite_2.2.0       backports_1.1.6     lattice_0.20-41     glue_1.4.0          digest_0.6.25      
[19] rvest_0.3.5         colorspace_1.4-1    pkgconfig_2.0.3     broom_0.5.6         haven_2.2.0         purrr_0.3.4        
[25] scales_1.1.0        processx_3.4.2      tibble_3.0.1        generics_0.0.2      ellipsis_0.3.0      withr_2.2.0        
[31] tidyverse_1.3.0     BiocGenerics_0.34.0 hexbin_1.28.1       cli_2.0.2           magrittr_1.5        crayon_1.3.4       
[37] readxl_1.3.1        memoise_1.1.0       ps_1.3.2            fs_1.4.1            ncdf4_1.17          fansi_0.4.1        
[43] nlme_3.1-147        forcats_0.5.0       xml2_1.3.2          pkgbuild_1.0.8      tools_4.0.0         prettyunits_1.1.1  
[49] hms_0.5.3           lifecycle_0.2.0     munsell_0.5.0       reprex_0.3.0        callr_3.4.3         compiler_4.0.0     
[55] rlang_0.4.6         grid_4.0.0          rstudioapi_0.11     testthat_2.3.2      gtable_0.3.0        codetools_0.2-16   
[61] DBI_1.1.0           R6_2.4.1            lubridate_1.7.8     knitr_1.28          bit_1.1-15.2        rprojroot_1.3-2    
[67] ProtGenerics_1.20.0 readr_1.3.1         desc_1.2.0          stringi_1.4.6       vctrs_0.2.4         dbplyr_1.4.3       
[73] tidyselect_1.0.0    xfun_0.13  

> # Read one example file. 
> rawfile <- file.path("./raw/20180220_14_autoQC01.raw")
> 
> #Tell me the peptide sequence you would like to extract peaks:
> pepSeq <- c("AGGSSEPVTGLADK", "GDLDAASYYAPVR", "FLLQFGAQGSPLFK")
> 
> #Calculate the tehoretical mass of M+2H based on peptide sequence:
> mass2Hplus <- (parentIonMass(pepSeq) + 1.008)/2
> 
> #Extract the selected peptides from the data file:
> XIC <- readXICs(rawfile = rawfile, masses = mass2Hplus, tol = 10)
> plot(XIC)
Error in xy.coords(x, y, xlabel, ylabel, log) : 
  'x' is a list, but does not have components 'x' and 'y'
>

cpanse, 2020-05-25:
@viviantra thanks for reporting. until we patch that you can use
```
plot.XICs(XIC)
```

viviantra, 2020-05-25:
That works.  Thank you!