fgcz/rawDiag #57: Split one plot into multiple pages

3H2ORZ, 2020-05-12:
Thanks for developing this great tool!
I have 20 raw files to read, plot and export as PDF. Some functions (e.g, cycle time plot) stack all these files together in one page of PDF, which are not readable in the end. Is there a solution to split one into several pages?
[image]

tobiasko, 2020-05-14:
Hi @3H2ORZ,

nice to hear that you like our tool! Are you using the Shiny app for plotting or the R command line?

Best,
Tobi

cpanse, 2020-05-14:
@tobiaskoI guess there is a ggplot option or function to fix that.

3H2ORZ, 2020-05-14:
Hi Tobi,

Thanks for reaching out. I am using the R command line to generate the pdf file. 

Best,
Runsheng.

ctrachse, 2020-05-19:
Hi @3H2ORZ 

have you tried the suggested solution? Did it work?

best 
Christian

3H2ORZ, 2020-05-19:
Hi Christian,

Thanks for the quick response. 
I tried it several times, but it looked like the function ScanFrequMovingOver cannot be called in my R. It reported: Error in ScanFrequMovingOver(.) :  could not find function "ScanFrequMovingOver".
I also tried once to find an explanation of this function by typing: ?ScanFrequMovingOver. The page opened successfully. Do I miss something installed together with the package? Any suggestions?

Best, 
RZheng.

closed 2020-05-21 by 3H2ORZ