fgcz/rawDiag #59: R 4.0.2 - Error: `x` must be a vector, not a `data.frame/rawDiag` object.

cpanse, 2020-07-15:
```
cp@fgcz-148:~ > R

R version 4.0.2 (2020-06-22) -- "Taking Off Again"
Copyright (C) 2020 The R Foundation for Statistical Computing
Platform: x86_64-pc-linux-gnu (64-bit)

R is free software and comes with ABSOLUTELY NO WARRANTY.
You are welcome to redistribute it under certain conditions.
Type 'license()' or 'licence()' for distribution details.

  Natural language support but running in an English locale

R is a collaborative project with many contributors.
Type 'contributors()' for more information and
'citation()' on how to cite R or R packages in publications.

Type 'demo()' for some demos, 'help()' for on-line help, or
'help.start()' for an HTML browser interface to help.
Type 'q()' to quit R.

[Previously saved workspace restored]

R> library(rawDiag)
Package 'rawDiag' version 0.0.34
R> ?read.raw
R> ?read.raw
R>  (rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))
[1] "/home/cp/R/x86_64-pc-linux-gnu-library/4.0/rawDiag/extdata/sample.raw"
     system.time(RAW <- read.raw(file = rawfile))
R>      system.time(RAW <- read.raw(file = rawfile))
executing mono /home/cp/R/x86_64-pc-linux-gnu-library/4.0/rawDiag/exec/fgcz_raw.exe /home/cp/R/x86_64-pc-linux-gnu-library/4.0/rawDiag/extdata/sample.raw qc ...
MasterScanNumber calculated
renamed LMmZCorrectionppm to LMCorrection
renamed AGCPSMode to PrescanMode
   user  system elapsed 
  0.411   0.043   0.442 
Warning messages:
1: In is.rawDiag(object) :
  missing column name(s): MasterScanNumber, LMCorrection, ElapsedScanTimesec, transient, AGCMode, PrescanMode
2: `funs()` is deprecated as of dplyr 0.8.0.
Please use a list of either functions or lambdas: 

  # Simple named list: 
  list(mean = mean, median = median)

  # Auto named with `tibble::lst()`: 
  tibble::lst(mean, median)

  # Using lambdas
  list(~ mean(., trim = .2), ~ median(., na.rm = TRUE))
This warning is displayed once every 8 hours.
Call `lifecycle::last_warnings()` to see where this warning was generated. 
R> PlotScanFrequency(RAW)
Error: `x` must be a vector, not a `data.frame/rawDiag` object.
Run `rlang::last_error()` to see where the error occurred.
R> 
```

cpanse, 2020-07-15:
solution 

```{r}
class(RAW) <- 'data.frame'
PlotScanFrequency(RAW)
```

Stortebecker, 2020-11-19:
Same problem here, using rawDiag 0.0.34 and R 3.6.2. 
Thanks for the workaround!

cpanse, 2020-11-19:
@Stortebecker please see also our latest release https://github.com/fgcz/rawDiag/releases/tag/0.0.38
and our new package:
https://github.com/fgcz/rawR
C