fgcz/rawDiag #60: Problems reading Meta Data from QExactive HF / HFX Instruments

BerndMaier, 2020-07-27:
First, thanks for your amazing package.
I have an issue with our newer instruments.

RawDiag 0.0.34 returns an Error when Using read.raw.info 

( > QE_HFX_Meta <- read.raw.info(QE_HFX)
Error in !silent : invalid argument type )

With one QExactive HF and one QExactive HFX Instrument.
Other Instruments (Fusion, Velos, EMR, UHMR and another QE HF) work fine.

Both instruments also control their Thermo Vanquish LC Systems through Xcalibur. 
All other instruments have separate computers to control the LC.

So maybe this causes a difference in data structure of the *.raw Files, as the traces (UV, Column Temp, Pump Preassure) are added as analog traces.

[image]

Thank you very much,
Bernd

tobiasko, 2020-07-27:
Hi Bernd,

so the MS instruments connected to the Vanquish LCs use Xcalibur & LCDevices or is the LC integration SII-based? Are the UV, column temp, pump pressure traces really going through the D/A converter of the MS? There is only two analog inputs available on the HF and the HF-X. We use the analog inputs to record the LC system pressure and %B, but we use 3rd party hardware (Waters LCs). I thought the Vanquish communication is SII-based and digital using the ethernet connection. Anyway: In the context of the raw file each detector generates an instrument stream. Analog detectors can also have channels (shared time axis), if the detector uses the same sampling rate.

We would need to check what exactly is returned by your files (stream & instrument structure).

BerndMaier, 2020-07-27:
Hi Tobias,
the LC is connected to the PC via USB connection - so given the screen shot I took from Xcalibur - i think it should be a SII (unfortunately) i don't know the term)
 
[image]

I sent you an Email with a link to Rawfiles from both instruments.

Thank you very much,
Bernd

tobiasko, 2020-07-27:
Yes, that looks like SII for Xcalibur. Hmmmm. Ok. So, all our LC-MS combinations use Xcalibur & LCDevices, so we have zero experience with this way of recording data into raw files. I expect that SII generates separate data streams for UV, column temp, pump pressure,... but we need to check your files.

I think our current code only uses detectors of the type `MS` or `MStrending`

BerndMaier, 2020-07-27:
Yeah, we are switching to FreeStyle - but old habbits ;)

But the installed Thermo Software shouldn't affect rawDiag and the functions ?

I really would be quite happy in accessing the data from read.raw.info as Sample Name, Column Name etc.  would be really useful for our Dashboards.

Thanks again,
Bernd

tobiasko, 2020-07-28:
@cpanse The raw files are available at `fgcz-r-033.uzh.ch:/scratch/tobiasko/rawDiagIssues/#60`

tobiasko, 2020-07-28:
Hi Bernd,

I checked your file, but can't reproduce your error message:

```
> rawDiag::read.raw.info(file = "HF_Buffer_A.raw")
$`Number of instruments`
[1] 4

$`General File Information`
$`General File Information`$`RAW file`
[1] "HF_Buffer_A.raw"

$`General File Information`$`RAW file version`
[1] 66

$`General File Information`$`Creation date`
[1] "7/16/2020 11:36:46 AM"

$`General File Information`$Operator
[1] "TEDEVELP"

$`General File Information`$`Number of instruments`
[1] 4

$`General File Information`$Description
NULL

$`General File Information`$`Instrument model`
[1] "Q Exactive HF Orbitrap"

$`General File Information`$`Instrument name`
[1] "Q Exactive HF Orbitrap"

$`General File Information`$`Serial number`
[1] "Exactive Series slot"

$`General File Information`$`Software version`
[1] "2.11-211101/2.11.0.3006"

$`General File Information`$`Firmware version`
[1] "rev. 1"

$`General File Information`$Units
[1] "None"

$`General File Information`$`Mass resolution`
[1] 0.5

$`General File Information`$`Number of scans`
[1] 24399

$`General File Information`$`Number of ms2 scans`
[1] 1037

$`General File Information`$`Scan range`
[1]     1 24399

$`General File Information`$`Time range`
[1]  0 59

$`General File Information`$`Mass range`
[1]   50 8000

$`Sample Information`
$`Sample Information`$`Sample name`
NULL

$`Sample Information`$`Sample id`
[1] "Eluent A"

$`Sample Information`$`Sample type`
[1] "Unknown"

$`Sample Information`$`Sample comment`
[1] "Eluent A"

$`Sample Information`$`Sample vial`
[1] "R:A1"

$`Sample Information`$`Sample volume`
[1] 0

$`Sample Information`$`Sample injection volume`
[1] 10

$`Sample Information`$`Sample row number`
[1] 0

$`Sample Information`$`Sample dilution factor`
[1] 1

$`Filter Information`
$`Filter Information`$`Scan filter (first scan)`
[1] "FTMS + p ESI Full ms [200.0000-2000.0000]"

$`Filter Information`$`Scan filter (last scan)`
[1] "FTMS + p ESI Full ms [200.0000-2000.0000]"

$`Filter Information`$`Total number of filters`
[1] 147

```

Could it be that your `file` argument is not set correctly?

tobiasko, 2020-07-28:
> sessionInfo()
R version 3.6.3 (2020-02-29)
Platform: x86_64-apple-darwin15.6.0 (64-bit)
Running under: macOS Catalina 10.15.4

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/3.6/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawDiag_0.0.34

loaded via a namespace (and not attached):
 [1] httr_1.4.1            tidyr_1.0.2           bit64_0.9-7          
 [4] jsonlite_1.6.1        splines_3.6.3         foreach_1.4.8        
 [7] modelr_0.1.6          gtools_3.8.1          assertthat_0.2.1     
[10] statmod_1.4.34        blob_1.2.1            cellranger_1.1.0     
[13] yaml_2.2.1            ggrepel_0.8.2         pillar_1.4.3         
[16] RSQLite_2.2.0         backports_1.1.5       lattice_0.20-40      
[19] glue_1.3.2            limma_3.42.2          doSNOW_1.0.18        
[22] digest_0.6.25         rvest_0.3.5           minqa_1.2.4          
[25] colorspace_1.4-1      preprocessCore_1.48.0 Matrix_1.2-18        
[28] plyr_1.8.6            pkgconfig_2.0.3       broom_0.5.5          
[31] haven_2.2.0           purrr_0.3.3           scales_1.1.0         
[34] snow_0.4-3            gdata_2.18.0          lme4_1.1-21          
[37] tibble_2.1.3          generics_0.0.2        ggplot2_3.3.0        
[40] tidyverse_1.3.0       hexbin_1.28.1         protViz_0.6.3        
[43] readxl_1.3.1          survival_3.1-11       magrittr_1.5         
[46] crayon_1.3.4          memoise_1.1.0         fs_1.3.2             
[49] nlme_3.1-145          MASS_7.3-51.5         gplots_3.0.3         
[52] xml2_1.3.1            forcats_0.5.0         tools_3.6.3          
[55] data.table_1.12.8     hms_0.5.3             minpack.lm_1.2-1     
[58] lifecycle_0.2.0       stringr_1.4.0         reprex_0.3.0         
[61] munsell_0.5.0         MSstats_3.18.5        compiler_3.6.3       
[64] caTools_1.18.0        rlang_0.4.5           grid_3.6.3           
[67] nloptr_1.2.2.1        iterators_1.0.12      rstudioapi_0.11      
[70] marray_1.64.0         bitops_1.0-6          boot_1.3-24          
[73] gtable_0.3.0          codetools_0.2-16      DBI_1.1.0            
[76] reshape2_1.4.3        R6_2.4.1              lubridate_1.7.8      
[79] dplyr_0.8.5           bit_1.1-15.2          readr_1.3.1          
[82] KernSmooth_2.23-16    stringi_1.4.6         parallel_3.6.3       
[85] Rcpp_1.0.4            vctrs_0.2.4           dbplyr_1.4.2         
[88] tidyselect_1.0.0

tobiasko, 2020-07-28:
same for the HF-X file:

```
> rawDiag::read.raw.info(file = "HFX_Buffer_A.raw")
$`Number of instruments`
[1] 4

$`General File Information`
$`General File Information`$`RAW file`
[1] "HFX_Buffer_A.raw"

$`General File Information`$`RAW file version`
[1] 66

$`General File Information`$`Creation date`
[1] "6/3/2020 11:06:07 AM"

$`General File Information`$Operator
[1] "TEDEVELP"

$`General File Information`$`Number of instruments`
[1] 4

$`General File Information`$Description
NULL

$`General File Information`$`Instrument model`
[1] "Q Exactive HF-X Orbitrap"

$`General File Information`$`Instrument name`
[1] "Q Exactive HF-X Orbitrap"

$`General File Information`$`Serial number`
[1] "Exactive Series slot"

$`General File Information`$`Software version`
[1] "2.11-211101/2.11.0.3006"

$`General File Information`$`Firmware version`
[1] "rev. 1"

$`General File Information`$Units
[1] "None"

$`General File Information`$`Mass resolution`
[1] 0.5

$`General File Information`$`Number of scans`
[1] 24413

$`General File Information`$`Number of ms2 scans`
[1] 500

$`General File Information`$`Scan range`
[1]     1 24413

$`General File Information`$`Time range`
[1]  0 55

$`General File Information`$`Mass range`
[1]   50 2150

$`Sample Information`
$`Sample Information`$`Sample name`
[1] "Eluent A"

$`Sample Information`$`Sample id`
NULL

$`Sample Information`$`Sample type`
[1] "Unknown"

$`Sample Information`$`Sample comment`
[1] "Eluent A"

$`Sample Information`$`Sample vial`
[1] "R:B1"

$`Sample Information`$`Sample volume`
[1] 0

$`Sample Information`$`Sample injection volume`
[1] 10

$`Sample Information`$`Sample row number`
[1] 0

$`Sample Information`$`Sample dilution factor`
[1] 0

$`Filter Information`
$`Filter Information`$`Scan filter (first scan)`
[1] "FTMS + p ESI Full lock ms [200.0000-2000.0000]"

$`Filter Information`$`Scan filter (last scan)`
[1] "FTMS + p ESI Full lock ms [200.0000-2000.0000]"

$`Filter Information`$`Total number of filters`
[1] 89

```

tobiasko, 2020-07-28:
```
> rawDiag::read.raw.info(file = "HFX_with_UV.raw")
$`Number of instruments`
[1] 6

$`General File Information`
$`General File Information`$`RAW file`
[1] "HFX_with_UV.raw"

$`General File Information`$`RAW file version`
[1] 66

$`General File Information`$`Creation date`
[1] "6/26/2020 5:25:03 AM"

$`General File Information`$Operator
[1] "TEDEVELP"

$`General File Information`$`Number of instruments`
[1] 6

$`General File Information`$Description
NULL

$`General File Information`$`Instrument model`
[1] "Q Exactive HF-X Orbitrap"

$`General File Information`$`Instrument name`
[1] "Q Exactive HF-X Orbitrap"

$`General File Information`$`Serial number`
[1] "Exactive Series slot"

$`General File Information`$`Software version`
[1] "2.11-211101/2.11.0.3006"

$`General File Information`$`Firmware version`
[1] "rev. 1"

$`General File Information`$Units
[1] "None"

$`General File Information`$`Mass resolution`
[1] 0.5

$`General File Information`$`Number of scans`
[1] 24937

$`General File Information`$`Number of ms2 scans`
[1] 6340

$`General File Information`$`Scan range`
[1]     1 24937

$`General File Information`$`Time range`
[1]  0 55

$`General File Information`$`Mass range`
[1]   50 8000

$`Sample Information`
$`Sample Information`$`Sample name`
[1] "1%B 0,5mL/min 70C Messung2"

$`Sample Information`$`Sample id`
NULL

$`Sample Information`$`Sample type`
[1] "Unknown"

$`Sample Information`$`Sample comment`
[1] "1%B 0,5mL/min 70C Messung2"

$`Sample Information`$`Sample vial`
[1] "B:A5"

$`Sample Information`$`Sample volume`
[1] 0

$`Sample Information`$`Sample injection volume`
[1] 10

$`Sample Information`$`Sample row number`
[1] 35

$`Sample Information`$`Sample dilution factor`
[1] 0

$`Filter Information`
$`Filter Information`$`Scan filter (first scan)`
[1] "FTMS + p ESI Full lock ms [200.0000-2000.0000]"

$`Filter Information`$`Scan filter (last scan)`
[1] "FTMS + p ESI Full ms [200.0000-2000.0000]"

$`Filter Information`$`Total number of filters`
[1] 2605

```

tobiasko, 2020-07-28:
What you can see is that the `*with_UV` file contains two additional instruments that recorded data.

BerndMaier, 2020-07-28:
First: Glad to hear that it should work properly. 

I again tried on 2 machines with R 3.63 and 4.02

Hope I don't have any stupid mistake in read.raw.info action -  but with all other instruments it works fine.

[image]
[image]

tobiasko, 2020-07-28:
What do you get when executing:

```
> (rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw'))
> read.raw.info(rawfile)
```
???

BerndMaier, 2020-07-28:
The same error ...

rawfile <- file.path(path.package(package = 'rawDiag'), 'extdata', 'sample.raw')
> read.raw.info(rawfile)
Error in !silent : invalid argument type

But with other instruments 

> Velos <- ("I://Rohdaten/Orbitrap/M/M200703/M20070308_SST.raw")
> Velos_Meta <- read.raw.info(Velos)
> Velos_Meta
$`Number of instruments`
[1] 2

$`General File Information`
$`General File Information`$`RAW file`
[1] "M20070308_SST.raw"

$`General File Information`$`RAW file version`
[1] 66

$`General File Information`$`Creation date`
[1] "7/3/2020 11:28:55 AM"

$`General File Information`$Operator
[1] "TEDEVELP"

$`General File Information`$`Number of instruments`
[1] 2

tobiasko, 2020-07-28:
That you can not process the package example data is pretty strange. I am having the feeling your error is not connected to specific files. What do you get:

```
> rawfile
[1] "/Library/Frameworks/R.framework/Versions/3.6/Resources/library/rawDiag/extdata/sample.raw"
> class(rawfile)
[1] "character"
> 
```

BerndMaier, 2020-07-28:
> rawfile
[1] "C:/Users/maierb5/Documents/R/win-library/4.0/rawDiag/extdata/sample.raw"
> class(rawfile)
[1] "character"

Took a look at Sample.raw
Its also a QExactive HF-X Instrument

tobiasko, 2020-07-29:
Hmmm...currently I have no idea how to proceed, especially because I can not reproduce your error on my system. Maybe @cpanse can help!?

BerndMaier, 2020-07-29:
Thanks for the efforts. 

I try to find a none Windows System to test. But at least here on 3 independent Windows machines I have the same problem.
As your test was on a Mac system. Don't know if this should make any difference

Just saw that the beginning of the function
___
> read.raw.info
function (file, mono = if (Sys.info()["sysname"] %in% c("Darwin", 
    "Linux")) TRUE else FALSE, exe = file.path(path.package(package = "rawDiag"), 
    "exec", "fgcz_raw.exe"), mono_path = "", 
    argv = "info", system2_call = TRUE, method = "thermo")
___

Makes a difference for mono (whatever this is)

Thank you very much again

tobiasko, 2020-07-29:
Yes, there is a difference. On Windows you just need to have the MS [.NET framework](https://dotnet.microsoft.com/download) installed.

According to [planet orbitrap](https://planetorbitrap.com/rawfilereader) you should have >4.5.1

[Mono](https://www.mono-project.com/) is the open source .NET implementation needed on MacOS to use the RawFileReader dlls. I hope at some point RawFileReader will be using .NETcore, but Thermo hasn't been the most progressive company in the past when it comes to software development.

tobiasko, 2020-07-29:
Finally! On a Windows system I get the error too:

```
> rawfile
[1] "C:/Program Files/R/R-4.0.0/library/rawDiag/extdata/sample.raw"
> read.raw.info(rawfile)
Error in !silent : invalid argument type
> sessionInfo()
R version 4.0.0 (2020-04-24)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server >= 2012 x64 (build 9200)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawDiag_0.0.34

loaded via a namespace (and not attached):
 [1] tidyselect_1.0.0 purrr_0.3.4      haven_2.2.0      lattice_0.20-41  colorspace_1.4-1
 [6] vctrs_0.2.4      generics_0.0.2   yaml_2.2.1       blob_1.2.1       rlang_0.4.5     
[11] hexbin_1.28.1    pillar_1.4.3     glue_1.4.0       DBI_1.1.0        tidyverse_1.3.0 
[16] bit64_0.9-7      dbplyr_1.4.3     protViz_0.6.4    modelr_0.1.6     readxl_1.3.1    
[21] lifecycle_0.2.0  stringr_1.4.0    munsell_0.5.0    gtable_0.3.0     cellranger_1.1.0
[26] rvest_0.3.5      codetools_0.2-16 memoise_1.1.0    forcats_0.5.0    parallel_4.0.0  
[31] fansi_0.4.1      broom_0.5.6      Rcpp_1.0.4.6     readr_1.3.1      scales_1.1.0    
[36] backports_1.1.7  jsonlite_1.6.1   farver_2.0.3     fs_1.4.1         bit_1.1-15.2    
[41] ggplot2_3.3.0    hms_0.5.3        digest_0.6.25    stringi_1.4.6    dplyr_0.8.5     
[46] grid_4.0.0       cli_2.0.2        tools_4.0.0      magrittr_1.5     tibble_3.0.0    
[51] RSQLite_2.2.0    crayon_1.3.4     tidyr_1.0.2      pkgconfig_2.0.3  ellipsis_0.3.0  
[56] xml2_1.3.1       reprex_0.3.0     lubridate_1.7.8  assertthat_0.2.1 httr_1.4.1      
[61] R6_2.4.1         nlme_3.1-147     compiler_4.0.0  

```

tobiasko, 2020-07-29:
@cpanse Don't we have a unit test for `read.raw.info()` that should fail? Or is the unit testing only done on MS OS?

BerndMaier, 2020-07-29:
Ok, it is good that the Error is reproducible on the developer side, too.

Have so much bad experience with MassSpecs doing crazy things ... until Thermo or Waters is on site and everything is back to normal... until they leave.

tobiasko, 2020-07-29:
I guess we will have a fix soon. I can't really tell you what is going wrong on Windows, but now we can follow the problem. :-)

tobiasko, 2020-08-04:
I discovered something:

```
> read.raw.info("~/Downloads/HF_Buffer_A.raw")
Error in !silent : invalid argument type
> setwd("~/Downloads")
> read.raw.info("HF_Buffer_A.raw")
$`Number of instruments`
[1] 4

$`General File Information`
$`General File Information`$`RAW file`
[1] "HF_Buffer_A.raw"

$`General File Information`$`RAW file version`
[1] 66

```

Reading from the current working directory is fine, using a rel. path fails.

BerndMaier, 2020-08-04:
Unfortunatelly not here - also no difference between network and local drive or R version R 3.6.3 // 4.0.2
___
setwd("I://Users/maierb5/Transfer/")
> read.raw.info(file = "HF_Buffer_A.raw")
Error in !silent : invalid argument type
> 
___
setwd("C://Users/xxxx/Downloads/")
> read.raw.info("HF_Buffer_A.raw")
Error in !silent : invalid argument type
> 
____

cpanse, 2020-08-20:
> > rawfile
> > [1] "C:/Users/maierb5/Documents/R/win-library/4.0/rawDiag/extdata/sample.raw"
> > class(rawfile)
> > [1] "character"
> 
> Took a look at Sample.raw
> Its also a QExactive HF-X Instrument
> 
> Just attached read.raw.info of our Thermo Instruments
> 
> [image]
> 
> [image]

@BerndMaier Can you give it a try using R(> 4.0)
```{r}
pkgs <- c('dplyr', 'hexbin', 'protViz', 'RSQLite', 'scales', 'tidyr', 'tidyverse', 'shiny') pkgs <- pkgs[(!pkgs %in%
unique(installed.packages()[,'Package']))] if(length(pkgs) > 0){install.packages(pkgs)} 

install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.38.tar.gz',
repo=NULL, type='source')
```

tobiasko, 2020-08-20:
```
> info <- read.raw.info("HFX_with_UV.raw")
> str(info)
List of 31
 $ RAW file                : chr "HFX_with_UV.raw"
 $ RAW file version        : chr "66"
 $ Creation date           : chr "6/26/2020 5:25:03 AM"
 $ Operator                : chr "TEDEVELP"
 $ Number of instruments   : num 6
 $ Description             : chr ""
 $ Instrument model        : chr "Q Exactive HF-X Orbitrap"
 $ Instrument name         : chr "Q Exactive HF-X Orbitrap"
 $ Serial number           : chr "Exactive Series slot #6063"
 $ Software version        : chr "2.11-211101/2.11.0.3006"
 $ Firmware version        : chr "rev. 1"
 $ Units                   : chr "None"
 $ Mass resolution         : chr "0.500"
 $ Number of scans         : num 24937
 $ Number of ms2 scans     : num 6340
 $ Scan range              : num [1:2] 1 24937
 $ Time range              : num [1:2] 0 55
 $ Mass range              : num [1:2] 50 8000
 $ Scan filter (first scan): chr "FTMS + p ESI Full lock ms [200.0000-2000.0000]"
 $ Scan filter (last scan) : chr "FTMS + p ESI Full ms [200.0000-2000.0000]"
 $ Total number of filters : chr "2605"
 $ Sample name             : chr "1%B 0,5mL/min 70C Messung2"
 $ Sample id               : chr ""
 $ Sample type             : chr "Unknown"
 $ Sample comment          : chr "1%B 0,5mL/min 70C Messung2"
 $ Sample vial             : chr "B:A5"
 $ Sample volume           : chr "0"
 $ Sample injection volume : chr "10"
 $ Sample row number       : chr "35"
 $ Sample dilution factor  : chr "0"
 $ Sample barcode          : chr ""
> sessionInfo()
R version 4.0.0 (2020-04-24)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server >= 2012 x64 (build 9200)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawDiag_0.0.38

loaded via a namespace (and not attached):
 [1] tidyselect_1.0.0 purrr_0.3.4      haven_2.2.0      lattice_0.20-41  colorspace_1.4-1
 [6] vctrs_0.2.4      generics_0.0.2   yaml_2.2.1       blob_1.2.1       rlang_0.4.5     
[11] hexbin_1.28.1    pillar_1.4.3     glue_1.4.0       DBI_1.1.0        tidyverse_1.3.0 
[16] bit64_0.9-7      dbplyr_1.4.3     modelr_0.1.6     readxl_1.3.1     protViz_0.6.4   
[21] lifecycle_0.2.0  stringr_1.4.0    munsell_0.5.0    gtable_0.3.0     cellranger_1.1.0
[26] rvest_0.3.5      codetools_0.2-16 memoise_1.1.0    forcats_0.5.0    parallel_4.0.0  
[31] fansi_0.4.1      broom_0.5.6      Rcpp_1.0.4.6     readr_1.3.1      backports_1.1.7 
[36] scales_1.1.0     jsonlite_1.6.1   fs_1.4.1         bit_1.1-15.2     ggplot2_3.3.0   
[41] hms_0.5.3        digest_0.6.25    stringi_1.4.6    dplyr_0.8.5      grid_4.0.0      
[46] cli_2.0.2        tools_4.0.0      magrittr_1.5     tibble_3.0.0     RSQLite_2.2.0   
[51] crayon_1.3.4     tidyr_1.0.2      pkgconfig_2.0.3  ellipsis_0.3.0   xml2_1.3.1      
[56] reprex_0.3.0     lubridate_1.7.8  assertthat_0.2.1 httr_1.4.1       R6_2.4.1        
[61] nlme_3.1-147     compiler_4.0.0  
```

BerndMaier, 2020-08-20:
Works like a charm - Thank you very much

I saw that you changed the structure of the Output to 1 big list - is this intentional - will stay this way for future releases ?

____

rawfile <- "I://Users/maierb5/Transfer/HFX_with_UV.raw"
> read.raw.info(rawfile)
$`RAW file`
[1] "HFX_with_UV.raw"

$`RAW file version`
[1] "66"

$`Creation date`
[1] "6/26/2020 5:25:03 AM"

$Operator
[1] "TEDEVELP"

$`Number of instruments`
[1] 6

$Description
[1] ""

$`Instrument model`
[1] "Q Exactive HF-X Orbitrap"

$`Instrument name`
[1] "Q Exactive HF-X Orbitrap"

$`Serial number`
[1] "Exactive Series slot #6063"

$`Software version`
[1] "2.11-211101/2.11.0.3006"

$`Firmware version`
[1] "rev. 1"

$Units
[1] "None"

$`Mass resolution`
[1] "0.500"

$`Number of scans`
[1] 24937

$`Number of ms2 scans`
[1] 6340

$`Scan range`
[1]     1 24937

$`Time range`
[1]  0 55

$`Mass range`
[1]   50 8000

$`Scan filter (first scan)`
[1] "FTMS + p ESI Full lock ms [200.0000-2000.0000]"

$`Scan filter (last scan)`
[1] "FTMS + p ESI Full ms [200.0000-2000.0000]"

$`Total number of filters`
[1] "2605"

$`Sample name`
[1] "1%B 0,5mL/min 70C Messung2"

$`Sample id`
[1] ""

$`Sample type`
[1] "Unknown"

$`Sample comment`
[1] "1%B 0,5mL/min 70C Messung2"

$`Sample vial`
[1] "B:A5"

$`Sample volume`
[1] "0"

$`Sample injection volume`
[1] "10"

$`Sample row number`
[1] "35"

$`Sample dilution factor`
[1] "0"

$`Sample barcode`
[1] ""