fgcz/rawDiag #61: Reading Analag Traces from RawFiles

BerndMaier, 2020-07-27:
Hi there,

2 of our instruments have their LC Systems (Thermo Vanquish) directly connected through Xcalibur, with the MassSpecs (QE HF & HFX), within one system.

The LC traces (UV, Column Temp, Pump pressure etc.) are written as analog traces to the MS rawfile.

Is there a possibility to read these analog traces with rawDiag - just as the TIC trace ?

[image]

Thank you very much,
Bernd

tobiasko, 2020-07-27:
Hi Bernd,

I think we haven't tried this so far, but according to the RawFileReader documentation it is possible to read data streams coming from other detectors types than `MS`.

We need to check your data.

Best,
Tobi

BerndMaier, 2020-07-27:
Thank you very much - just added a File where we recorded UV to the Raw Files

tobiasko, 2020-07-28:
@cpanse Could you try to implement a function that wraps the example C# code for `Instrument (detector) List` shown in `UsingRawFileReader` as well as `Getting detector details`???

tobiasko, 2020-07-30:
Regarding analog traces: I checked a raw file that we recorded on a Exploris 480 with both A/D inputs enabled. We feed two signals from our LCs into these ports. Using the following code you can already access the recordings and plot as a function of time:

```
> autoQC01 <- read.raw(file = "20200730_001_autoQC01.raw", rawDiag = FALSE)
system2 is writting to tempfile  C:\Users\TOBIAS~1.FGC\AppData\Local\Temp\2\RtmpWcTL9L\file1b682c1c1f3tsv ...
unlinking C:\Users\TOBIAS~1.FGC\AppData\Local\Temp\2\RtmpWcTL9L\file1b682c1c1f3tsv ...
> xyplot(AnalogInAV ~ StartTime, data = autoQC01)
```

[image]

I am not sure if all Thermo systems log using `AnalogInAV` and `AnalogInBV`. The same data should be shown in Freestyle when you display a Chromatogram Range for the detector `MS Trending`.

tobiasko, 2020-07-30:
Your analog traces seem to be empty:

```
> HFX_with_UV <- read.raw(file = "HFX_with_UV.raw", rawDiag = FALSE)
system2 is writting to tempfile  C:\Users\TOBIAS~1.FGC\AppData\Local\Temp\2\RtmpWcTL9L\file1b68321f3cb2tsv ...
unlinking C:\Users\TOBIAS~1.FGC\AppData\Local\Temp\2\RtmpWcTL9L\file1b68321f3cb2tsv ...
> summary(HFX_with_UV$AnalogInput1V)
   Min. 1st Qu.  Median    Mean 3rd Qu.    Max. 
      0       0       0       0       0       0 
```

tobiasko, 2020-07-30:
Most likely your LC-related data is logged as additional instrument. Currently, we don't have a function that allows accessing other instruments, but we will check if a code extension is possible.

BerndMaier, 2020-07-31:
Thank you very much for checking

Would be a really nice option - as to date Thermo itself has no option to quickly extract the UV traces in an automated way.