fgcz/rawDiag #62: Read additional Infos from Raw Files (Status Log, Tune Method) etc.

BerndMaier, 2020-07-27:
Hi,
is there a possibility to access some additional information of the MS raw files.

Like the Status Log - containing the instrument read outs (voltages, temperature, pressure etc.) for every scan - accessible usually through Xcalibur -> view -> report -> status log

[image]

Thank you very much,
Bernd

tobiasko, 2020-07-27:
Hi Bernd,

yes, there is a lot of scan attributes already available through rawDiag, please check:

https://github.com/fgcz/rawDiag#343--how-to-get-all-scan-attributes-assosiated-to-each-scan

You should see a similar list in Freestyle when selecting `Chromatogram ranges -> Detector type: MS Trending`

[image]

Best,
Tobi

BerndMaier, 2020-08-03:
Thanks Tobi,
I checked the parameters in read.raw(...., rawDiag =F) and read.raw.info with some Fusion Files

Any chance to obtain the information in Reports\Sample Information ? Couldn't find them in the long long list read.raw outputs and read.raw.info just has some of them. 

I wanted to use them to store some meta data information (column, project, sample matrix etc.) and read them automated when samples are finished.  
And it would be easier for the end-user if everything has its own column, than posting it comma separated in e.g. Comment

[image]

___
test[["Sample Information"]]
$`Sample name`
[1] "SST"

$`Sample id`
NULL

$`Sample type`
[1] "Unknown"

$`Sample comment`
[1] "A04RCL18043"

$`Sample vial`
NULL

$`Sample volume`
[1] 0

$`Sample injection volume`
[1] 0

$`Sample row number`
[1] 1

$`Sample dilution factor`
[1] 1
___

Thank you very much again,
Bernd

tobiasko, 2020-08-04:
Hi Bernd,

not sure if I get your question correctly (comma separated in comment???) `read.raw.info` returns a nested list.

```
> info <- read.raw.info("HF_Buffer_A.raw")
> class(info)
[1] "list"
> str(info)
List of 4
 $ Number of instruments   : int 4
 $ General File Information:List of 18
  ..$ RAW file             : chr "HF_Buffer_A.raw"
  ..$ RAW file version     : int 66
  ..$ Creation date        : chr "7/16/2020 11:36:46 AM"
  ..$ Operator             : chr "TEDEVELP"
  ..$ Number of instruments: int 4
  ..$ Description          : NULL
  ..$ Instrument model     : chr "Q Exactive HF Orbitrap"
  ..$ Instrument name      : chr "Q Exactive HF Orbitrap"
  ..$ Serial number        : chr "Exactive Series slot"
  ..$ Software version     : chr "2.11-211101/2.11.0.3006"
  ..$ Firmware version     : chr "rev. 1"
  ..$ Units                : chr "None"
  ..$ Mass resolution      : num 0.5
  ..$ Number of scans      : int 24399
  ..$ Number of ms2 scans  : int 1037
  ..$ Scan range           : int [1:2] 1 24399
  ..$ Time range           : num [1:2] 0 59
  ..$ Mass range           : num [1:2] 50 8000
 $ Sample Information      :List of 9
  ..$ Sample name            : NULL
  ..$ Sample id              : chr "Eluent A"
  ..$ Sample type            : chr "Unknown"
  ..$ Sample comment         : chr "Eluent A"
  ..$ Sample vial            : chr "R:A1"
  ..$ Sample volume          : int 0
  ..$ Sample injection volume: int 10
  ..$ Sample row number      : int 0
  ..$ Sample dilution factor : int 1
 $ Filter Information      :List of 3
  ..$ Scan filter (first scan): chr "FTMS + p ESI Full ms [200.0000-2000.0000]"
  ..$ Scan filter (last scan) : chr "FTMS + p ESI Full ms [200.0000-2000.0000]"
  ..$ Total number of filters : int 147
```
Please also check `?read.raw.info`. I am also not sure what you are missing. Since the nested lists are named you can even select list entries by name:

```
> info$`Sample Information`$`Sample id`
[1] "Eluent A"
```

tobiasko, 2020-08-04:
Does this help? You might find [this](http://adv-r.had.co.nz/Data-structures.html) helpful.

BerndMaier, 2020-08-04:
Sorry for the confusion.

Yeah, i know about the returned structure. (But funny - I'm going through AdrancedR at the moment :) )

I posted the returned Sample Information of our Velos above.

I'm missing some information like Laboratory, Company, Phone etc. that are accessible from QualBrowser and most likely Freestyle.
___
These are all Columns you could use at Xcalibur Sequence creation

[image]
___
And we thought we could use the other columns for projects, columns, sample matrices, digestion enzymes etc. and gather the information through read.raw.info automatically (as users are lazy and will forget to upload their sequences manually)

The Comma Separation was just for a workaround if the Phone ... etc would not be accessible (as we could put the information above all into the Comment Field with _**,**_ or _**;**_ and Split it afterwards with StringR) but it would be much nicer if everything would have its own column at sequence creating in Xcalibur.

Sorry again for the confusion and thanks for your help,
Bernd

tobiasko, 2020-08-04:
Ahhhh! ok. Now I get it! @cpanse Could we add the other variables (Laboratory, Company, Phone, ...). I assume we already have access, but didn't include them due to a lack of personal interest.

BerndMaier, 2020-08-04:
Thanks - I also didn't care about these Xcalibur-Columns in the last 15 years - just now as we take "Data Topics" a bit more serious, we try to gather some additional information, to make statistics or just filter these criteria in Shiny and generate HTML-Reports for e.g. a specific column and or all 4h Tryptic digests etc.   -  thats where they now become useful for us.

BerndMaier, 2021-02-05:
Hi there, saw these columns accessible in rawrr :+1:  
Will this be added to rawDiag, too?

Not sure if this question makes sense here: Is rawDiag also still in development - or just rawrr ?

cpanse, 2021-02-05:
@BerndMaier in the future we will run rawrr under the hood of rawDiag. If we find a way to make you happy we will give it a try.

BerndMaier, 2021-02-05:
@cpanse  oh - i'm already very happy to have now access through rawrr::readFileHeader - just wanted to understand how the two packages fit together - :1st_place_medal:  thanks again