fgcz/rawDiag #64: Cannot find dependency "Spectra"

joerg-b, 2020-07-31:
When trying to install rawDiag I get this error:

ERROR: dependency ‘Spectra’ is not available for package ‘rawDiag’

I cannot find a package of this name on CRAN or Bioconductor. Where can I get it?

BerndMaier, 2020-07-31:
Already tried to install like this ?

https://github.com/fgcz/rawDiag/releases

Think I got the same error message with the Version 0.0.36 from another page.

Spectra belongs to:
https://www.rformassspectrometry.org/pkgs/

tobiasko, 2020-08-06:
Hi @joerg-b,

which version of rawDiag are you using?

Our latest release is [Proteomics Forum 2019](https://github.com/fgcz/rawDiag/releases), or version 0.0.34. This release does not depend on `Spectra`. We hope to compile a new release soon.

Best,
Tobi

closed 2020-10-06 by tobiasko