fgcz/rawDiag #66: Installation of new RawFileReader files

breidan, 2020-09-28:
I would like to use `rawDiag` but do not find installation instructions at Thermo as suggested in the README file. The only thing I could find seems to involve VisualStudio which is not installed on the PC supposed to run `rawDiag`. I cannot install VisualStudio because the PC in question is not connected to the internet because of company policies. 

How can I make `rawDiag` run now? Any suggestions?

breidan, 2020-09-30:
Since no one answered yet I want to be a bit more concrete. The help of `read.raw` says:  
> this function is a generic adapter function for reading mass spectrometric measurement. It requieres the New RawFileReader dll files
>ThermoFisher.CommonCore.BackgroundSubtraction.dll,
>ThermoFisher.CommonCore.MassPrecisionEstimator.dll
>ThermoFisher.CommonCore.RawFileReader.dll
>be installed on the system. This can be done by using the Global Assembly Cache (GAC) or by setting the MONO_PATH enviroment.

Does this mean that for a WIN7 box they have to be in the GAC or would MONO_PATH also work? And if MONO_PATH would work which files are expected: only the DLLs or others as well?

breidan, 2020-10-01:
In the meantime I've decided to install the WIN version of `mono`, set `mono=TRUE` and MONO_PATH to the location of the dlls and `read.raw` works fine.

tobiasko, 2020-10-05:
Hi @breidan,

I am sorry to hear that no one answered your request. Could you please let me know which OS you are using, which version of rawDiag and R? The easiest is to use `sessionInfo()`

breidan, 2020-10-05:
Hi @tobiasko 

Here is the `sessioninfo()`:
>R version 3.6.1 (2019-07-05)
>Running under: Windows 7 x64 (build 7601) Service Pack 1
>Matrix products: default
>
>locale:
>[1] LC_COLLATE=English_United Kingdom.1252  LC_CTYPE=English_United Kingdom.1252   
>[3] LC_MONETARY=English_United Kingdom.1252 LC_NUMERIC=C                           
>[5] LC_TIME=English_United Kingdom.1252    
>
>attached base packages:
>[1] stats4    parallel  stats     graphics  grDevices utils     datasets  methods   base     
>
>other attached packages:
> [1] PCAtools_1.2.0      cowplot_1.0.0       lattice_0.20-38     reshape2_1.4.4      ggrepel_0.8.2      
> [6] ggplot2_3.3.2       xcms_3.10.1         MSnbase_2.12.1      ProtGenerics_1.18.0 S4Vectors_0.24.4   
>[11] mzR_2.20.0          Rcpp_1.0.5          BiocParallel_1.20.1 Biobase_2.46.0      BiocGenerics_0.32.0
>[16] MUtools_1.1         magrittr_1.5        enviPat_2.4         nlme_3.1-140       
>
>loaded via a namespace (and not attached):
> [1] BiocSingular_1.2.2          vsn_3.54.0                  foreach_1.5.0              
> [4] DelayedMatrixStats_1.8.0    assertthat_0.2.1            BiocManager_1.30.10        
> [7] affy_1.64.0                 dqrng_0.2.1                 GenomeInfoDbData_1.2.2     
>[10] robustbase_0.93-6           impute_1.60.0               pillar_1.4.6               
>[13] glue_1.4.2                  limma_3.42.2                digest_0.6.25              
>[16] GenomicRanges_1.38.0        RColorBrewer_1.1-2          XVector_0.26.0             
>[19] colorspace_1.4-1            preprocessCore_1.48.0       Matrix_1.2-17              
>[22] plyr_1.8.6                  MALDIquant_1.19.3           XML_3.99-0.3               
>[25] pkgconfig_2.0.3             zlibbioc_1.32.0             purrr_0.3.4                
>[28] scales_1.1.1                RANN_2.6.1                  affyio_1.56.0              
>[31] tibble_3.0.3                generics_0.0.2              IRanges_2.20.2             
>[34] ellipsis_0.3.1              withr_2.2.0                 SummarizedExperiment_1.16.1
>[37] cli_2.0.2                   MassSpecWavelet_1.52.0      crayon_1.3.4               
>[40] ncdf4_1.17                  fansi_0.4.1                 doParallel_1.0.15          
>[43] MASS_7.3-52                 tools_3.6.1                 lifecycle_0.2.0            
>[46] matrixStats_0.56.0          stringr_1.4.0               munsell_0.5.0              
>[49] irlba_2.3.3                 DelayedArray_0.12.3         pcaMethods_1.78.0          
>[52] compiler_3.6.1              GenomeInfoDb_1.22.1         rsvd_1.0.3                 
>[55] mzID_1.24.0                 rlang_0.4.7                 grid_3.6.1                 
>[58] RCurl_1.98-1.2              iterators_1.0.12            rstudioapi_0.11            
>[61] bitops_1.0-6                gtable_0.3.0                codetools_0.2-16           
>[64] R6_2.4.1                    dplyr_1.0.2                 stringi_1.4.6              
>[67] vctrs_0.3.2                 DEoptimR_1.0-8              tidyselect_1.1.0           

But why would that be of importance? The question was do I need to install VS to be able to run `rawDiag::read.raw` because the RawFileReader DLLs must be located in the GAC? Using MONO on WIN solved this and makes me feel better because I am using OpenSource.

tobiasko, 2020-10-06:
There is no need to install mono on a Windows OS, since Microsoft provides the .NET libraries. Mono is an open source implementation of Microsoft's .NET Framework for none MS Windows systems. Have you checked if .NET was installed on your system?

https://dotnet.microsoft.com/

tobiasko, 2020-10-06:
please check https://github.com/fgcz/rawDiag#11-net-framework-and-r

breidan, 2020-10-06:
Dear @tobiasko,

I did read the documentation. I did install .NET (V4.7.02558). I did get the RawFileReader assemblies. Unfortunately, the documentation does not provide any guidance of what to do with the RawFileReader assemblies. The only hint is the help for `read.raw` where it says they are expected to be in the GAC. The only information about how to place assemblies in the GAC is from MS saying that one needs to install VisualStudio to obtain the tools to manipulate the GAC. I cannot, and also do not want to, install VS.
So as a work-around I installed Mono, and now can use MONO_PATH to point to the assemblies. 

Maybe it would be beneficial to give some advise about what to do with the RawFileReader assemblies. That would save a number of people a lot of time searching.

tobiasko, 2020-10-06:
Dear @breidan,

I very much agree with you that our section https://github.com/fgcz/rawDiag#12-the-new-rawfilereader-net-assembly-from-thermo-fisher-scientific kind of diverts the problem back to the source which is not ideal. @cpanse and me already discussed this topic many times and I always voted that we should ship those .NET assemblies with the package to prevent frustration. Unfortunately, he had some legal concerns. But... to me knowledge the latest `rawDiag` release now ships with the assemblies. 

Have you tried [slim](https://github.com/fgcz/rawDiag/releases/tag/0.0.40).?

I couldn't find `rawDiag` in your `sessionInfo()` output.

breidan, 2020-10-06:
@tobiasko ,
I did try slim and also the Proteomics Forum 2019 release. Both did not work. `read.raw` would complain about not finding the RawFileReader dlls. And when you look at the slim tar ball then the dlls are not contained. 
I removed `rawDiag` because I was testing different approaches and wanted to start clean every time.

cpanse, 2020-10-06:
@breidan 

```
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.38.tar.gz', repo=NULL)
```

requires R (>= 4.0) or higher from your `sessionInfo` I see that you use R 3.6.

breidan, 2020-10-06:
@cpanse 
That is true. I took the liberty of manipulating the DESCRIPTION file so it would install with R 3.6. And it did install just fine.

cpanse, 2020-10-06:
OK; Have you tried to terminate R and start it again?

tobiasko, 2020-10-06:
> @tobiasko ,
> I did try slim and also the Proteomics Forum 2019 release. Both did not work. `read.raw` would complain about not finding the RawFileReader dlls. And when you look at the slim tar ball [...]

I just downloaded the tar.gz archive using a web browser and checked the content:

[image]

The archive contains the RawFileReader dlls

breidan, 2020-10-06:
@tobiasko 

If I download [https://github.com/fgcz/rawDiag/archive/0.0.40.tar.gz](url) and unpack with 7-zip then there are no dlls in the exec dir.

tobiasko, 2020-10-06:
You need to download http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.38.tar.gz
Sorry for the confusing numbering! The slim release is version 0.0.40, but corresponding archive is labeled `0.0.38`

breidan, 2020-10-06:
@tobiasko, @cpanse,

This is confusing. I downloaded http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.38.tar.gz, changed the DESCRIPTION and installed on R3.6. It installed ok, the dlls are present, and `read.raw` is working just fine now. 
As I said above: the documentation should be brought up to speed. 
Thanks for the help.

cpanse, 2020-10-06:
we released 0.0.38. tnx

breidan, 2020-10-09:
I just downloaded  http://fgcz-ms.uzh.ch/~cpanse/rawDiag_0.0.38.tar.gz to install on a linux box with R4.0. But it did not install because the new RawFileReader assemblies are missing. What happened to a few days ago when I downloaded and they were there?

closed 2020-11-17 by breidan